espressif/esp-now #96: DMD32.h and save data

hcetky, 2023-10-09:
Hi,
I'm using a relay to detect power outages in a project, and I want to detect the interruption and save the count when the power goes out. I want to save it when the low-power pin is low. At the same time, I'm trying to display the count on a P10 panel using the DMD32.h library. 
The system works independently of each other, but it doesn't work when I combine them. There are no code errors, etc. When I comment out "timerAlarmEnable(timer);", counting and saving works, but the P10 display doesn't work. When I enable the code, I get the following error. Thanks everybody.

i am using ESP32-Dev Module

Code:
`
#include <DMD32.h>
#include "fonts/droid_sans_12.h"
#include <Preferences.h>
#include <ESPAsyncWebServer.h>
#include <WiFi.h>

Preferences preferences;

#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1

DMD dmd(DISPLAYS_ACROSS, DISPLAYS_DOWN);

hw_timer_t * timer = NULL;

int counter;
int lowpower = 13;

void IRAM_ATTR triggerScan() {
  dmd.scanDisplayBySPI();
}
AsyncWebServer server(80);

void setup() {
  Serial.begin(9600);

  preferences.begin("saveone", false);
  counter = preferences.getUInt("saveone",0); 

  pinMode(lowpower, INPUT);
   
  uint8_t cpuClock = ESP.getCpuFreqMHz();
  timer = timerBegin(0, cpuClock, true);
  delay(300);
  timerAttachInterrupt(timer, &triggerScan, true);
  delay(300);
  timerAlarmWrite(timer, 600, true);
  delay(300);
  timerAlarmEnable(timer);
  delay(300);
}

void loop() {
    if (digitalRead(lowpower) == HIGH){
      preferences.putUInt("saveone", counter); 
      delay(1000);
    }
}

void yazdirSayi() {
  dmd.selectFont(Droid_Sans_12);
  dmd.clearScreen(true);

  int birler = counter % 10;
  int onlar = (counter / 10) % 10;
  int yuzler = (counter / 100) % 10;
  int binler = (counter / 1000) % 10;

  dmd.drawChar(5, 3, binler + '0', GRAPHICS_NORMAL);
  dmd.drawChar(11, 3, yuzler + '0', GRAPHICS_NORMAL);
  dmd.drawChar(17, 3, onlar + '0', GRAPHICS_NORMAL);
  dmd.drawChar(23, 3, birler + '0', GRAPHICS_NORMAL);
}`

Error and Restart Message:

`
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4008f024  PS      : 0x00060d35  A0      : 0x8008dcc2  A1      : 0x3ffb1e40  
A2      : 0x3ffb6e98  A3      : 0x3ffb81a4  A4      : 0x00000004  A5      : 0x00060d23  
A6      : 0x00060d23  A7      : 0x00000001  A8      : 0x3ffb81a4  A9      : 0x00000001  
A10     : 0x3ffb81a4  A11     : 0x00000001  A12     : 0x3ffb1f78  A13     : 0x0000b028  
A14     : 0x007bf578  A15     : 0x003fffff  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400846cd  LEND    : 0x400846d5  LCOUNT  : 0x00000027  

Backtrace: 0x4008f021:0x3ffb1e40 0x4008dcbf:0x3ffb1e60 0x4008cefe:0x3ffb1e80 0x40105caf:0x3ffb1ec0 0x40105d99:0x3ffb1ef0 0x400827ed:0x3ffb1f10 0x40082661:0x3ffb1f40 0x4008266b:0x3ffb1f60 0x40088d5d:0x3ffb1f80 0x4008231d:0x3ffb1fa0 0x400e22b7:0x3ffb1ff0 0x400e3cf2:0x3ffb2010 0x400e42b6:0x3ffb2030 0x400e4466:0x3ffb2�
`

lhespress, 2023-10-10:
@hcetky You compile the code based on arduino, right? I think it's better to posted a placeholder request for open discussion in the [arduino-esp32](https://github.com/espressif/arduino-esp32/issues).

hcetky, 2023-10-12:
ok, Thank you, I temporarily resolved the issue by disabling the timer alarm as follows before saving it, I fixed the error. Thanks to everyone.

Solution:
timerAlarmDisable(timer); // I added this code before using preferences.
preferences.putUInt("saveone", counter);

closed 2023-10-12 by hcetky