espressif/esp-now #100: wireless_debug example web server vue project sources

cotestatnt, 2023-11-09:
Would it be possible to obtain the Vue source files with which the web server included in this example was implemented?
It would be really appreciated
Thank you very much!

[wireless_debug](https://github.com/espressif/esp-now/tree/master/examples/wireless_debug/components/web_server)

lhespress, 2023-11-10:
@cotestatnt Please see te link as follow: 
https://github.com/espressif/esp-now/blob/master/examples/wireless_debug/components/web_server/web_server.c

cotestatnt, 2023-11-11:
Hi @lhespress 
Thank you for your reply.
Anyway I meant the sources for the webserver's page itself, not for the ESP32 firmware.

Here we have only the [compiled front-end](https://github.com/espressif/esp-now/tree/master/examples/wireless_debug/components/web_server/front), but the Vue sources are missing

lhespress, 2024-02-20:
@cotestatnt Please reference the attachment file: 
[归档.zip](https://github.com/espressif/esp-now/files/14340524/default.zip)

closed 2024-04-10 by lhespress