espressif/esp-now #102: current master example-OTA issue

Connor-WangK, 2023-12-17:
OS: macos 14.2 apple-m2
SDK: both 4.4 5.1 5.2
run the example-OTA always fail
[image]

when I try esp-now ver1.0, everything running ok.

Connor-WangK, 2024-01-04:
nothing update...

lhespress, 2024-01-25:
@andy-danieal Could you check it again on the master branch?

andy-danieal, 2024-02-23:
We have tested with latest branch getting same issue.

ESP-IDF Version:- ESP-IDF v5.2-beta2-190-g4ecc978bd6
ESPNow Example:- e422fbc6267db141aeb22cdf3703f676b6efe75a

[ota_initiator_log.txt](https://github.com/espressif/esp-now/files/14381930/ota_master_log.txt)
[ota_responder_log.txt](https://github.com/espressif/esp-now/files/14381932/ota_responder_log.txt)
[sdkconfig_initator.txt](https://github.com/espressif/esp-now/files/14381942/sdkconfig_initator.txt)

Connor-WangK, 2024-02-23:
Just now, I do a test, everything is running ok.
env: macbook air m2,
sdk: 97594d2076cb2a2358062c9848b3491371c89a57 (HEAD -> release/v5.2)
chip: esp32c3

lhespress, 2024-02-23:
@Connor-WangK Thanks for your feedback, please close the issue if resolved.

andy-danieal, 2024-02-23:
@Connor-WangK  and @lhespress We have tried many times and also create new setup and update ESP-IDF with latest branch but getting same issue.

env: Ubuntu
sdk: ESP-IDF v5.2-166-ge0944287f3 
chip: esp32s2

closed 2024-02-23 by Connor-WangK

Connor-WangK, 2024-02-24:
maybe you can try with esp32c3, 
I'm no going on a business, can't do any tests with other chips but esp32c3.

ScarsFun, 2024-02-25:
Tested with esp32c3 and esp32s2. Issue persist.

andy-danieal, 2024-02-27:
@lhespress , I have tested with esp32s2 with CONFIG_ESPNOW_OTA_RETRANSMISSION_TIMES=2 then works.