espressif/esp-now #104: esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, wifi_event_handler, NULL) returns ESP_ERR_INVALID_STATE

Josh-TapNoa, 2024-01-15:
Hi all, 

Ive been working on a project that uses the 2.3.0 version of ESPNOW in its component form, as well as a private library based on ESPNOWs base .a file (Im sure this is bad practice, but I just dont have bandwidth to deal with that as is). 

When I attempt to deinit this base ESPNOW variant (henceforth localComs) and init the 2.3 version from components, I get an error suggesting that esp_event_handler_register within espnow_init is returning ESP_ERR_INVALID_STATE. 

`I (386774) Network: Sending OTA Request to controller!
Creating Hub Local OTA Task! 
Starting Test Task 
W (386776) ESP-NOW (Abstract): Deinit all
I (386780) ESPNOW: espnow [version: 1.0] deinit
LOCAL COMS IS LEAVING DEINIT! 
E (387286) Local Coms OTA - Common: Initting espnow from OTA Update Init!
I (387286) espnow: esp-now Version: 2.3.0
I (387287) ESPNOW: espnow [version: 1.0] init
I (387291) espnow: mac: 68:67:25:25:5f:5a, version: 2
I (387297) espnow: Enable main task
I (387301) Local Coms OTA - Common: OTA Init!
Local OTA Init!I (387307) espnow: main task entry
I (387806) ESPNOW: espnow [version: 1.0] deinit
I (387806) Local Coms OTA - Common: OTA Deinit!
I (387807) espnow: main task exit
I (388307) ESPNOW: espnow [version: 1.0] init
Local coms is leaving Init!I (388809) ESP-NOW (Abstract): Connecting to paired
I (388809) ESP-NOW (Abstract): Adding peer with unique mac address
Reinit on local coms!I (388813) ESP-NOW (Abstract): Using low power heartbeat timeout
I (388821) ESP-NOW (Abstract): Waiting for connection from paired device (hid 4)...
I (391689) wifi_metrics: rssi:-44 min_rssi_ever:-50
I (391800) ESP-NOW (Abstract): Sending message to addon to say hub ready
I (391800) ESP-NOW (Abstract): Sending message from hid 0 to hid 4
I (391804) ESP-NOW (Abstract): Sending message to mac: 48:27:e2:51:a9:da
I (391813) ESP-NOW (Abstract): Mac address matches hid 4
I (391816) ESP-NOW (Abstract): Paired device connected (HID 4, Chan 0)
I (391900) Network: Sending OTA Request to controller!
Creating Hub Local OTA Task! 
Starting Test Task 
W (391900) ESP-NOW (Abstract): Deinit all
I (391905) ESPNOW: espnow [version: 1.0] deinit
LOCAL COMS IS LEAVING DEINIT! 
E (392411) Local Coms OTA - Common: Initting espnow from OTA Update Init!
I (392411) espnow: esp-now Version: 2.3.0
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40092bdc
0x40092bdc: espnow_init at /workspaces/Xarkle-hub/components/esp-now/src/espnow/src/espnow.c:1034 (discriminator 1)

file: "./components/esp-now/src/espnow/src/espnow.c" line 1021
func: espnow_init
expression: esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, wifi_event_handler, NULL)

abort() was called at PC 0x4002c66b on core 0
0x4002c66b: _esp_error_check_failed at /opt/esp/idf/components/esp_system/esp_err.c:50

Backtrace: 0x4002546a:0x3ffff710 0x4002c675:0x3ffff730 0x400330a5:0x3ffff750 0x4002c66b:0x3ffff7c0 0x40092bdc:0x3ffff7f0 0x400919d3:0x3ffff830 0x4008ca1d:0x3ffff880 0x4002e862:0x3ffff8a0
0x4002546a: panic_abort at /opt/esp/idf/components/esp_system/panic.c:452

0x4002c675: esp_system_abort at /opt/esp/idf/components/esp_system/port/esp_system_chip.c:84

0x400330a5: abort at /opt/esp/idf/components/newlib/abort.c:38

0x4002c66b: _esp_error_check_failed at /opt/esp/idf/components/esp_system/esp_err.c:50

0x40092bdc: espnow_init at /workspaces/Xarkle-hub/components/esp-now/src/espnow/src/espnow.c:1034 (discriminator 1)

0x400919d3: ota_update_init at /workspaces/Xarkle-hub/components/ota_update/src/local_coms_ota_common.c:47

0x4008ca1d: hub_local_ota_test_task at /workspaces/Xarkle-hub/main/network.c:530

0x4002e862: vPortTaskWrapper at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162`

Just wondering if anyone has any advice (Other than ditching the "localComs" component) for dealing with such errors? 

Technically the localComs and the espnow 2.3 use the same espnow.a reference, so it should be possible to use them safely within the same project?

Josh-TapNoa, 2024-01-15:
Also the internal xQueue for ESPNOW is init at the head of the task? Surely you'd expect the queue to be init in init? Theres technically a very small period of time, where espnow believes itself init, but in actual fact the queue hasnt been init yet?

lhespress, 2024-01-25:
@Josh-TapNoa Please attachment your project code for show how to reproduce the issue.

closed 2024-04-10 by lhespress