espressif/esp-now #107: ESP-NOW won't build for target esp32c6

Teesmo, 2024-01-25:
OS - Ubuntu 23.10
IDF - release/v5.1
Development board - ESP32-C6-DevkitM-1 v1.0

When I try to build code for the example: 

> esp-now/examples/control 

I get the error :
```
error: 'wifi_pkt_rx_ctrl_t' {aka 'esp_wifi_rxctrl_t'} has no member named 'secondary_channel'
  263 |              rx_ctrl->secondary_channel, espnow_data->size, espnow_data->payload,
      |                     ^~
```
I have pulled the latest version of my release of the IDF (v5.1), and have also updated the submodules.

lhespress, 2024-01-26:
@Teesmo Please git apply the patch as the attachment and check again.
[support_esp32c6.zip](https://github.com/espressif/esp-now/files/14059755/support_esp32c6.zip)

lhespress, 2024-01-31:
@Teesmo The master branch has supported it, please update and check again.

Teesmo, 2024-02-01:
I had already tried the patch, which worked, but I'll now update as recommended, thanks.

closed 2024-04-10 by lhespress