espressif/esp-now #108: Is it possible to send OTA files via UART without internet in ESPNow?

andy-danieal, 2024-01-29:
**ESPNow gateway** was connected via UART to the **Ubuntu system** and **many devices** (with different types of firmware) were connected to the gateway to communicate with ESPNow.

*Notes*:- ESPNow gateways have different firmware from other devices.

n3b0j5a, 2024-01-30:
Yeah, you can do that, you need to develop some kind of protocol to download firmware via UART (to some dedicated or OTA partition), and than send it over via espnow.

andy-danieal, 2024-01-31:
**ESPNow Gateway** and **Device** firmware are different, will ESPNow Gateway act as an initiator?

Is there any example of downloading firmware via UART?

lhespress, 2024-02-01:
@andy-danieal 

Yeath, ESP-NOW gateway can store data for OTA.

You can reference the link as follows which is the serial protocol used by the UART:
https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/serial-protocol.html

closed 2024-02-27 by andy-danieal