espressif/esp-now #109: unable to run coin cell demo switch on ESP32

albrownwood, 2024-02-12:
Using ESP-IDF v5.1.2, ESP32, compilation and flashing without issue, I get the following text on boot and nothing happens:

```
Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory /Users/aaa/dev/esp-now/examples/coin_cell_demo/switch
Executing "/Users/aaa/.espressif/python_env/idf5.1_py3.11_env/bin/python /Users/aaa/dev/esp-idf/tools/idf_monitor.py -p /dev/cu.usbserial-0001 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 /Users/aaa/dev/esp-now/examples/coin_cell_demo/switch/build/switch.elf -m '/Users/aaa/.espressif/python_env/idf5.1_py3.11_env/bin/python' '/Users/aaa/dev/esp-idf/tools/idf.py'"...
--- esp-idf-monitor 1.3.4 on /dev/cu.usbserial-0001 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1264
load:0x40078000,len:15080
load:0x40080400,len:4
0x40080400: _init at ??:?

ho 8 tail 4 room 4
load:0x40080404,len:3164
entry 0x400805fc

```
The _get-started_ example executes correctly on the same prototype board (ESP devkit).

lhespress, 2024-02-19:
@albrownwood Please see the [switch's](https://github.com/espressif/esp-now/blob/master/examples/coin_cell_demo/README.md#switch) instructions. You need set `Use coin cell button` of example to `N` to simulate.

albrownwood, 2024-02-20:
@lhespress yes I did compile with Use coin cell button = N and the error mentioned was present.

lhespress, 2024-02-21:
@albrownwood 
The log level of switch use `CONFIG_LOG_DEFAULT_LEVEL_NONE=y` as default, so there isn't any LOG output. you can check the function by the LOG of bulb:

```
I (5161) app_bulb: bind, uuid: 08:f9:e0:1f:9d:34, initiator_type: 513
I (17901) app_bulb: app_bulb_ctrl_data_cb, initiator_attribute: 513, responder_attribute: 1, value: 1
I (19171) app_bulb: app_bulb_ctrl_data_cb, initiator_attribute: 513, responder_attribute: 1, value: 0
I (28411) app_bulb: app_bulb_ctrl_data_cb, initiator_attribute: 513, responder_attribute: 1, value: 1
I (29931) app_bulb: app_bulb_ctrl_data_cb, initiator_attribute: 513, responder_attribute: 1, value: 0
I (31281) app_bulb: app_bulb_ctrl_data_cb, initiator_attribute: 513, responder_attribute: 1, value: 1
```
Of course, please change it to `CONFIG_LOG_DEFAULT_LEVEL_INFO=y` if you want to show the LOG of switch, then you will see the LOG of switch as follows:

```
I (3981) app_switch: switch bind press
I (4000) espnow_ctrl: src_addr: 24:0a:c4:03:f5:14, espnow_ctrl_initiator_ack, channel: 1
I (6251) app_switch: switch send press
I (6289) espnow_ctrl: src_addr: 24:0a:c4:03:f5:14, espnow_ctrl_initiator_ack, channel: 1
I (9891) app_switch: switch send press
I (9911) espnow_ctrl: src_addr: 24:0a:c4:03:f5:14, espnow_ctrl_initiator_ack, channel: 1
I (12196) app_switch: switch send press
I (12267) espnow_ctrl: src_addr: 24:0a:c4:03:f5:14, espnow_ctrl_initiator_ack, channel: 1
```

closed 2024-04-10 by lhespress