espressif/esp-now #112: OTA example don't works with 2.5.0 ESP-NOW Component

ScarsFun, 2024-02-23:
I'm using idf (5.3.0)
going back to 2.4.0 ESP-NOW Component it works.

initiator log:

...
I (6378) esp_netif_handlers: example_netif_sta ip: 192.168.1.9, mask: 255.255.255.0, gw: 192.168.1.1                
I (6379) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.1.9                        
I (6387) example_common: Connected to example_netif_sta                                                             
I (6393) example_common: - IPv4 address: 192.168.1.9,                                                               
I (6398) example_common: - IPv6 address: fe80:0000:0000:0000:a276:4eff:fe76:b92c, type: ESP_IP6_ADDR_IS_LINK_LOCAL  
I (6410) wifi:Set ps type: 0, coexist: 0                                                                            
                                                                                                                    
I (6413) espnow: esp-now Version: 2.5.0                                                                             
I (6418) ESPNOW: espnow [version: 1.0] init                                                                         
I (6422) espnow: mac: a0:76:4e:76:b9:2c, version: 2                                                                 
I (6427) espnow: Enable main task                                                                                   
I (6431) espnow: main task entry                                                                                    
I (6432) app_main: Open HTTP connection: http://192.168.1.5:8070/blink.bin                                          
I (9641) app_main: The service download firmware is complete, Spend time: 3s                                        
I (9641) esp_image: segment 0: paddr=001f0020 vaddr=3c020020 size=0b3f8h ( 46072) map                               
I (9651) esp_image: segment 1: paddr=001fb420 vaddr=3fc8c800 size=01350h (  4944)                                   
I (9655) esp_image: segment 2: paddr=001fc778 vaddr=40380000 size=038a0h ( 14496)                                   
I (9665) esp_image: segment 3: paddr=00200020 vaddr=42000020 size=1a480h (107648) map                               
I (9682) esp_image: segment 4: paddr=0021a4a8 vaddr=403838a0 size=08ddch ( 36316)                                   
W (12202) app_main: espnow wait ota num: 1                                                                          
I (12202) espnow_ota_initatior: [espnow_ota_initiator_send, 394]: total_size: 209584, packet_num: 1008              
I (12272) espnow_ota_initatior: count: 0, Upgrade_initiator_send, requested_num: 1, unfinished_num: 1, successed_nu0
W (27122) espnow_ota_initatior: [espnow_ota_request_status, 293]: <ESP_OK> wait_ticks: 1000                         
W (27623) espnow_ota_initatior: [espnow_ota_request_status, 293]: <ESP_OK> wait_ticks: 500                          
W (28124) espnow_ota_initatior: [espnow_ota_request_status, 293]: <ESP_OK> wait_ticks: 500                          
W (28124) espnow_ota_initatior: ESP_ERR_ESPNOW_OTA_DEVICE_NO_EXIST                                                  
W (28130) app_main: [app_firmware_send, 150]<ERROR> espnow_ota_initiator_send                                       
I (28136) main_task: Returned from app_main() 

Responder log:

...
I (6337) esp_netif_handlers: example_netif_sta ip: 192.168.1.14, mask: 255.255.255.0, gw: 192.168.1.1               
I (6337) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.1.14                       
I (6345) example_common: Connected to example_netif_sta                                                             
I (6351) example_common: - IPv4 address: 192.168.1.14,                                                              
I (6357) example_common: - IPv6 address: fe80:0000:0000:0000:1291:a8ff:fe36:6728, type: ESP_IP6_ADDR_IS_LINK_LOCAL  
I (6368) wifi:Set ps type: 0, coexist: 0                                                                            
                                                                                                                    
I (6372) espnow: esp-now Version: 2.5.0                                                                             
I (6376) ESPNOW: espnow [version: 1.0] init                                                                         
I (6380) espnow: mac: 10:91:a8:36:67:28, version: 2                                                                 
I (6386) espnow: Enable main task                                                                                   
I (6390) espnow: main task entry                                                                                    
I (6391) main_task: Returned from app_main()                                                                        
W (77546) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1        
W (77549) espnow: [espnow_recv_cb, 345] Send event queue failed                                                     
W (80546) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1        
W (80552) espnow: [espnow_recv_cb, 345] Send event queue failed                                                     
W (83546) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1        
W (83552) espnow: [espnow_recv_cb, 345] Send event queue failed                                                     
W (86546) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1        
W (86552) espnow: [espnow_recv_cb, 345] Send event queue failed                                                     
W (89546) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1        
W (89552) espnow: [espnow_recv_cb, 345] Send event queue failed                                                     
W (92546) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1        
W (92552) espnow: [espnow_recv_cb, 345] Send event queue failed                                                     
W (95546) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1        
W (98546) espnow: [espnow_recv_cb, 345] Send event queue failed                                                     
W (98547) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1        
W (101546) espnow: [espnow_recv_cb, 345] Send event queue failed                                                    
I (101546) wifi:bcn_timeout,ap_probe_send_start                                                                     
W (104547) espnow: [espnow_recv_cb, 345] Send event queue failed                                                    
W (104547) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1       
W (107547) espnow: [espnow_recv_cb, 345] Send event queue failed                                                    
W (107553) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1       
W (110547) espnow: [espnow_recv_cb, 345] Send event queue failed                                                    
W (110553) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1       
W (113547) espnow: [espnow_recv_cb, 345] Send event queue failed                                                    
W (113553) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1       
W (116547) espnow: [espnow_recv_cb, 345] Send event queue failed                                                    
W (116553) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1       
W (119547) espnow: [espnow_recv_cb, 345] Send event queue failed                                                    
W (119553) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1       
W (122547) espnow: [espnow_recv_cb, 345] Send event queue failed                                                    
W (122553) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_FAIL> esp_now_send, channel: 1       
W (122553) espnow: [espnow_send_forward, 891]: [espnow_send_forward, 891] <ESP_ERR_ESPNOW_NO_MEM> esp_now_send, cha1
      

best regards

lhespress, 2024-02-23:
@ScarsFun It's a duplicate/cross-posting of https://github.com/espressif/esp-now/issues/102. please check again as @Connor-WangK provided commit.

andy-danieal, 2024-02-23:
@lhespress, I have tested with esp-now 2.4.0 with working but issue when used 2.5.0

lhespress, 2024-02-26:
@andy-danieal @ScarsFun The OTA enable forward as default from 2.5.0, please change `CONFIG_ESPNOW_OTA_RETRANSMISSION_TIMES=2` and check again.

ScarsFun, 2024-02-26:
@andy-danieal

closed 2024-02-26 by ScarsFun

andy-danieal, 2024-02-27:
@lhespress , It's works. Could you please explain why this changes require?

lhespress, 2024-02-27:
@andy-danieal If the number of retransmission times is too large, [It'll](https://github.com/espressif/esp-now/blob/master/src/espnow/src/espnow.c#L876 ) take a long time because of the forwarding function is enabled. Therefore, the data thrown by the [receive callback](https://github.com/espressif/esp-now/blob/master/src/espnow/src/espnow.c#L344) to the queue cannot be processed in time, so the queue is full and cause OTA failure.