espressif/esp-now #115: esp_now latency

programmeddeath1, 2024-03-30:
I am sending multiple values to a slave esp using esp_now broadcast. I am sending points at a frequency of 1 every 30ms and above. For the first few points the slave receives the message within 5 to 7ms. but for certain subsequent points it goes upto to 50ms and even 100ms

[esp32 speedtests.xlsx](https://github.com/espressif/esp-now/files/14812156/esp32.speedtests.xlsx)

Point | SentTime | ReceivedTime | ExecutedTime | SlaveReceivedTime | SlaveExecutedTime | CommunicationTime | TotalExecutionTime | SlaveCommunicationTime | SlaveTotalExecutionTime
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
5 | 1711798622.65379 | 1711798622.66174 | 1711798622.66286 | 1711798622.67827 | 1711798622.67946 | 7.95 | 9.07 | 15.41 | 16.6
6 | 1711798622.69424 | 1711798622.70244 | 1711798622.70363 | 1711798622.72855 | 1711798622.7297 | 8.2 | 9.39 | 24.93 | 26.07
7 | 1711798622.73481 | 1711798622.74294 | 1711798622.74411 | 1711798622.78723 | 1711798622.78854 | 8.13 | 9.3 | 43.13 | 44.43
8 | 1711798622.77513 | 1711798622.78358 | 1711798622.78472 | 1711798622.83769 | 1711798622.83887 | 8.46 | 9.59 | 52.97 | 54.16
9 | 1711798622.81556 | 1711798622.82401 | 1711798622.82515 | 1711798622.88931 | 1711798622.8906 | 8.45 | 9.59 | 64.15 | 65.45
10 | 1711798622.85593 | 1711798622.86431 | 1711798622.86551 | 1711798622.94036 | 1711798622.94153 | 8.38 | 9.57 | 74.86 | 76.03
11 | 1711798622.89636 | 1711798622.90404 | 1711798622.90528 | 1711798622.99113 | 1711798622.99247 | 7.69 | 8.92 | 85.86 | 87.19
12 | 1711798622.9367 | 1711798622.945 | 1711798622.94616 | 1711798623.04216 | 1711798623.04331 | 8.31 | 9.46 | 96.0 | 97.16
13 | 1711798622.97704 | 1711798622.98467 | 1711798622.98596 | 1711798623.09191 | 1711798623.09301 | 7.63 | 8.92 | 105.96 | 107.06
14 | 1711798623.01738 | 1711798623.02585 | 1711798623.02706 | 1711798623.14438 | 1711798623.14561 | 8.47 | 9.68 | 117.32 | 118.55
15 | 1711798623.05784 | 1711798623.06586 | 1711798623.06703 | 1711798623.19467 | 1711798623.1958 | 8.02 | 9.19 | 127.64 | 128.77

Above is a sample of the values
The max latency is gong upto 200ms and avg of 50 ms over 37 points sent.

What could be causing this latency in between? Is there a threshold for difference between intermediate points?

lhespress, 2024-04-03:
@programmeddeath1 Please see the test report as the attachment. 
[ESP-NOW Test Report.pdf](https://github.com/espressif/esp-now/files/14846115/ESP-NOW.Test.Report.pdf)

nanshenwei, 2024-05-11:
Here is my setup. 
Hope this helps.
https://github.com/nanshenwei/espnowSpeedTest.git

closed 2024-06-27 by lhespress