espressif/esp-now #116: Minor modifications to esp-idf espnow example causes intermittent crashes

0xjmux, 2024-03-31:
Hi all,
I'm trying to ingest messages from the Wizmote (which uses ESP-IDF to send messages) into an esp-idf program as part of a larger project. Something similar has been done before in [the WLED project](https://github.com/Aircoookie/WLED/blob/main/wled00/remote.cpp), albeit using arduino. 

I modified the `espnow` example to remove the sending code and to parse the data in the wizmote format, and it works - some of the time. For unknown reasons, after receiving a few (it's not usually the same amount) packets, the program will panic. I've seen a few errors, but the most common  has been a stack overflow, which makes me think it's likely a memory issue. I've been trying to use JTAG with an ESP32-S3 to debug it, but because of FreeRTOS's threading having a useful backtrace on a crash is a bit of a coin toss. Where I've been able to locate a crash back to in my code, I've left a comment indicating so. 

I've gone about as far as I can, but I'm tearing my hair out trying to figure out what's causing the crash, and none of the heap tracing or core dump features of the SDK have been working for 

If you'd be able to take a quick look at my test program and see if you can spot any obvious errors, I'd greatly appreciate it. 

Thanks!

#### Code

 
<details>
<summary>Expand here for code!</summary>

To duplicate my project structure, create an instance of the `espnow` example program in ESP-IDF, add the two files to the directory, and comment out `app_main()` from the original `espnow_example_main.c`.

espnow_remote_test.c

```c
/**
 * Test code for ingesting messages from the Wizmote ESP-NOW remote
 * espnow_remote_test.c
*/

// this include block copied from espnow_example_main.c
#include <stdlib.h>
#include <time.h>
#include <string.h>
#include <assert.h>
#include "freertos/FreeRTOS.h"
#include "freertos/semphr.h"
#include "freertos/timers.h"
#include "nvs_flash.h"
#include "esp_random.h"
#include "esp_event.h"
#include "esp_netif.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "esp_mac.h"
#include "esp_now.h"
#include "esp_crc.h"

#include "esp_app_trace.h"

#include "espnow_remote_test.h"

////////////////////////////////////////
// TEMPORARY DEFS
#define ESPNOW_WIFI_MODE WIFI_MODE_STA

static uint8_t s_example_broadcast_mac[ESP_NOW_ETH_ALEN] = { 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF };
// static uint16_t s_example_espnow_seq[EXAMPLE_ESPNOW_DATA_MAX] = { 0, 0 };

////////////////////////////////////////

static const char *TAG = "espnow_wizmote_example";

static uint32_t last_msg_seq = 0;      // seq number of last message
// static espnow_msg_structure incoming;           // holds incoming message data
static QueueHandle_t s_example_espnow_queue;    // semaphore for espnow handling

/* WiFi should start before using ESPNOW */
static void example_wifi_init(void)
{
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
[... 304 lines cut ...]
    uint8_t *data;
    int data_len;
    example_espnow_event_id_t id;
} example_espnow_event_recv_cb_t;

enum {
    EXAMPLE_ESPNOW_DATA_BROADCAST,
    EXAMPLE_ESPNOW_DATA_UNICAST,
    EXAMPLE_ESPNOW_DATA_MAX,
};

// FUNCTIONS

static void example_wifi_init(void);

uint8_t example_espnow_data_parse(uint8_t *data, uint16_t data_len, uint8_t *program, \
    uint32_t *seq, uint8_t *button);
static void espnow_recv_task(void *pvParameter);
static esp_err_t espnow_remote_recv_init(void);
void espnow_remote_recv_deinit(void);

#endif
```

</details>

0xjmux, 2024-04-01:
I forgot to include an example of a stack trace, so here's one of the more helpful ones I got: 
[image]

I really want to see the internals of `esp_now_is_peer_exist()`, but I can't seem to find it anywhere - I doubt that the issue is there and not in my code, but it'd still be helpful to have increased visibility into what its doing. 

~~I've attempted to use the "Start Heap Trace" feature in ESP-IDF - the button in the vscode extension doesn't give me any result under "Application Trace Archives", and as far as I can see no files are created in the repository directory. When I run it manually, (with `heap_trace_start()` as the first line of `example_espnow_recv_cb()` and `heap_trace_stop()` and dump at the end of `espnow_recv_task()`, this is the result I get:~~
After heap debugging in SDKCONFIG like so:
[image]

I no longer get the messages about crashes - I just get a stack overflow crash. After a lot of trying I was finally able to extract an .elf core dump that can be opened with `idf.py coredump-debug -c crash-coredump.elf`. In the interest of speeding up the debugging process, I've attached it to this comment. If someone would be able to help me with taking a look at it, I'd greatly appreciate it. Thanks!

[crash-coredump.elf.zip](https://github.com/espressif/esp-now/files/14827440/crash-coredump.elf.zip)

0xjmux, 2024-04-02:
After *many* more hours of attempted debugging, not much progress has been made. I've spent nearly 6 hours at this point trying to get the App and heap debugging features in ESP-IDF to work as intended, have read all the docs I can find, and have tried everything i can think of. My attempts to use [SystemView](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/app_trace.html#app-trace-system-behaviour-analysis-with-segger-systemview) have completely failed, and no matter what I do (including attaching an ESP-Prog to get an additional UART for just the core dump) nothing works. The ESP crashes and just stops printing data, and GDB gives me not super helpful backtraces like those seen above. 

Whenever I run the App/heap traces the log files end up empty; I have the libraries included in my code and have tried manually init'ing the functionality. I've tried both through the CLI and through the VScode extension - neither work. 

This is becoming extremely frustrating - If I can't even get minor modifications to the provided example to work without causing disastrous overflows, how am I supposed to be confident about using it in production?

<details>
<summary>Here's an info_coredump I was able to extract. This is about all the information I've been able to extract from the chip since the debugging tools refuse to work properly</summary>

```
Executing action: coredump-info
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3fcb2048, name: 'espnow_recv_tas', GDB name: 'process 1070276680'
Panic reason: ***ERROR*** A stack overflow in task espnow_recv_tas has been detected.

================== CURRENT THREAD REGISTERS ===================
exccause       0x1d (StoreProhibitedCause)
excvaddr       0x0
epc1           0x4037b2a3
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
pc             0x40375759          0x40375759 <panic_abort+33>
lbeg           0x400570e8          1074098408
lend           0x400570f3          1074098419
lcount         0x0                 0
sar            0x9                 9
ps             0x60f23             397091
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x8037c97c          -2143827588
a1             0x3fcb1760          1070274400
a2             0x3fcb17ac          1070274476
a3             0x3fcb17f3          1070274547
a4             0x3fc99118          1070174488
a5             0x3fcb1860          1070274656
a6             0x3fca5a44          1070225988
a7             0xffffffff          -1
a8             0x0                 0
a9             0x3fcb1720          1070274336
a10            0x0                 0
a11            0x0                 0
a12            0x4037b0f8          1077391608
a13            0x0                 0
a14            0xff                255
a15            0x60021             393249

==================== CURRENT THREAD STACK =====================
[... 76 lines cut ...]
.flash.rodata 0x3c080120 0x1fa78 RW A
.ext_ram.bss 0x3c0a0000 0x0 RW  
.iram0.data 0x40387100 0x0 RW  
.iram0.bss 0x40387100 0x0 RW  
.dram0.heap_start 0x3fca1a58 0x0 RW  
.coredump.tasks.data 0x3fcaaa24 0x154 RW 
.coredump.tasks.data 0x3fcaa750 0x2b0 RW 
.coredump.tasks.data 0x3fcb2048 0x154 RW 
.coredump.tasks.data 0x3fcb16a0 0x990 RW 
.coredump.tasks.data 0x3fca5890 0x154 RW 
.coredump.tasks.data 0x3fca55f0 0x280 RW 
.coredump.tasks.data 0x3fca71c4 0x154 RW 
.coredump.tasks.data 0x3fca6ec0 0x2e0 RW 
.coredump.tasks.data 0x3fca81a4 0x154 RW 
.coredump.tasks.data 0x3fca7ec0 0x2c0 RW 
.coredump.tasks.data 0x3fca3848 0x154 RW 
.coredump.tasks.data 0x3fca35a0 0x290 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```
</details>

Any help I can get resolving this issue as soon as possible would be greatly appreciated. Thanks.

lhespress, 2024-04-03:
@0xjmux Could you add your application code based on [get-started](https://github.com/espressif/esp-now/tree/master/examples/get-started) example and check again? BTW, [esp-now](https://github.com/espressif/esp-now/tree/master) component as an application-level [ESP-NOW](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html) which provides some high-level functionalities. In short, [espnow_xxx](https://github.com/espressif/esp-now/blob/master/src/espnow/include/espnow.h) adds some application processing based on [esp_now_xxx](https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_now.h). The detailed introduction, please see the link as follows:
https://github.com/espressif/esp-now/blob/master/User_Guide.md#introduction

0xjmux, 2024-04-03:
@lhespress I had tried to get the `get-started` example working previously - no matter what I do, it refuses to compile. I've followed those instructions before, but just did so again after freshly cloning the repository again. 

When I run `idf.py menuconfig` or `idf.py set-target esp32s3`, I get:
```
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Target changed from esp32 to esp32s3, solving dependencies.
Using component placed at /home/jmux/Github/Neopixel-Matrix-Tetris-Project/libs-examples/esp-now-main-repo for dependency esp-now(2.*), specified in /home/jmux/Github/Neopixel-Matrix-Tetris-Project/libs-examples/esp-now-main-repo/examples/get-started/main/idf_component.yml
....Updating lock file at /home/jmux/Github/Neopixel-Matrix-Tetris-Project/libs-examples/esp-now-main-repo/examples/get-started/dependencies.lock
Processing 3 dependencies:
[1/3] esp-now (2.5.0)
[2/3] espressif/cmake_utilities (0.5.3)
[3/3] idf (5.1.2)
CMake Error at /home/jmux/Documents/esp/esp-idf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'esp-now'.
Call Stack (most recent call first):
  /home/jmux/Documents/esp/esp-idf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /home/jmux/Documents/esp/esp-idf/tools/cmake/build.cmake:595 (__build_expand_requirements)
  /home/jmux/Documents/esp/esp-idf/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:14 (project)
```

When I run `idf.py build`, I get the same error. Trying to configure the project or compile in VSCode gives the same result. This is on a fresh clone of the repo with `idf.py --version` == ESP-IDF v5.1.2

lhespress, 2024-04-03:
@0xjmux It seems don't find `esp-now` components. Which environment do you used? Linux, Mac or Windows? BTW, the attachment is the firmware on my Mac environment.
[get-start.zip](https://github.com/espressif/esp-now/files/14848083/get-start.zip)

0xjmux, 2024-04-05:
I'm on Ubuntu 22.04LTS and ESP-IDF v5.1.2

It seems to find it fine in the example built in to ESP-IDF, and when I import `esp_now.h` into my program - but none of the examples in the esp-now examples folder will compile.

0xjmux, 2024-04-06:
I just attempted to upgrade my ESP-IDF to v5.2.1, and was quickly reminded why I had been avoiding doing that. The SDK still refuses to work with intellisense out of the box, and getting the bare minimum functional is quite the frustrating ordeal. If this is something you guys somehow aren't aware of, you should add testing the installation of ESP-IDF through VSCode to the list - fresh install, open any example, and you get a sea of red. I've lost count of how many hours I've spent just trying to get the SDK working - not being able to tell it to use a pyenv virtualenv during installation is another annoyance. 

It also didn't help, building the `get-started` example through the IDE or via `idf.py build` still fails with the same error.

0xjmux, 2024-04-07:
I've been losing my mind as to what I've been doing wrong, so as a sanity check I enabled comprehensive heap detection and tracing in the default `wifi/espnow` example, compiled it and ran it. 

It seems that I'm not going insane - it detects a stack overflow in the example that comes with ESP-IDF. 

idf.py openocd
```
Info : [esp32s3.cpu0] Target halted, PC=0x40375A81, debug_reason=00000001
Info : [esp32s3.cpu0] Halt cause (***ERROR*** A stack overflow in task example_espnow_ has been detected.)
```

idf.py monitor: 
```
I (18574) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (18574) espnow_example: Receive error data from: [MAC OF MY OTHER ESPNOW DEVICE]
[at this point, it crashes and nothing else shows up]
```

idf.py gdbtui
```
[esp32s3.cpu0] Target halted, PC=0x40375A81, debug_reason=00000001
[esp32s3.cpu0] Halt cause (***ERROR*** A stack overflow in task example_espnow_ has been detected.)
[New Thread 1070275316]
[New Thread 1070245204]
[New Thread 1070230808]
[New Thread 1070234848]

Thread 8 "example_espnow_" received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 1070275316]
0x40375a81 in panic_abort (details=0x3fcb12ac "***ERROR*** A stack overflow in task example_espnow_ has been detected.") at /home/jmux/Documents/esp/v5.2.1/
esp-idf/components/esp_system/panic.c:472
(gdb) bt
#0  0x40375a81 in panic_abort (details=0x3fcb12ac "***ERROR*** A stack overflow in task example_espnow_ has been detected.")
    at /home/jmux/Documents/esp/v5.2.1/esp-idf/components/esp_system/panic.c:472
#1  0x4037c6f4 in esp_system_abort (details=0x3fcb12ac "***ERROR*** A stack overflow in task example_espnow_ has been detected.")
    at /home/jmux/Documents/esp/v5.2.1/esp-idf/components/esp_system/port/esp_system_chip.c:93
#2  0x4037d5c1 in vApplicationStackOverflowHook (xTask=0x3fcb1af4, pcTaskName=0x3fcb1b28 "example_espnow_")
    at /home/jmux/Documents/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:553
#3  0x4037e9f2 in vTaskSwitchContext () at /home/jmux/Documents/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3630
#4  0x4037d687 in _frxt_dispatch () at /home/jmux/Documents/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
#5  0x4037d67d in _frxt_int_exit () at /home/jmux/Documents/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:24
```

I created a brand new project from the `wifi/espnow` example, and haven't made any modifications other than enabling heap debugging features in menuconfig. 

I should mention that this is on the latest stable release of ESP-IDF, since I bit the bullet and spent 4 hours reconfiguring the extension to work again after updating it. 
```
$ idf.py --version
ESP-IDF v5.2.1
```

Since this bug is not in my modified code but rather in the example code that ships with ESP-IDF, it should be easier to triage from your guy's end. If someone could take a look at resolving that, I'd greatly appreciate it.

zhangyanjiaoesp, 2024-04-07:
@0xjmux I have tested the `wifi/espnow` example on the IDF v5.2.1 on my side, when the `comprehensive heap detection` option is enabled, it does generate a task watchdog. 
When the `comprehensive heap detection` option is enabled, it will consume more memory, so you need to increase the stack size of the `example_espnow_task`. After I increase the stack size to 4096, the example can work well.

> xTaskCreate(example_espnow_task, "example_espnow_task", 4096, send_param, 4, NULL);

For your original code, you can also increase the stack size of the watchdog task, and then enable `comprehensive heap detection` option to debug.

closed 2024-06-27 by lhespress