espressif/esp-now #122: Issue with ota responder

augustpevko, 2024-06-28:
It seems that [this](https://github.com/espressif/esp-now/blob/e8748dd33eced44e5f0dd6ad7ba11a39e4ab71d9/src/ota/espnow_ota_responder.c#L292) static variable might not be resetting properly. If you call this function after the variable exceeds 100 percent, it won't compare correctly in [this](https://github.com/espressif/esp-now/blob/e8748dd33eced44e5f0dd6ad7ba11a39e4ab71d9/src/ota/espnow_ota_responder.c#L302) if statement. As a result, no storage write occurs, and no event is posted.

lhespress, 2024-07-01:
@augustpevko Could you provide the detailed steps on how to reproduce the variable exceeds 100 percent?

augustpevko, 2024-07-01:
@lhespress You should be able to reproduce it by using [this code](https://gist.github.com/augustpevko/593cdc9cdc407da492dbfe9b9663ff15) in the OTA example. Change the `urls` variable to include two different firmware files to load.

lhespress, 2024-07-04:
@augustpevko Thanks for you shared the reproduce code. Please apply the patch 
[espnow_ota_responder.zip](https://github.com/user-attachments/files/16096810/espnow_ota_responder.zip)
and check again. 
```
I (21914) espnow_ota_responder: The device starts to upgrade
I (26657) app_main: Progress of OTA update: 10%
I (30405) app_main: Progress of OTA update: 20%
I (33787) app_main: Progress of OTA update: 30%
I (37585) app_main: Progress of OTA update: 40%
I (40888) app_main: Progress of OTA update: 50%
I (44320) app_main: Progress of OTA update: 60%
I (48413) app_main: Progress of OTA update: 70%
I (51802) app_main: Progress of OTA update: 80%
I (55015) app_main: Progress of OTA update: 90%
I (58518) app_main: Progress of OTA update: 100%
I (58518) espnow_ota_responder: Write total_size: 172736, written_size: 172736, spend time: 36s
I (58523) espnow_ota_responder: Running firmware version: v1.0-202-ge8748dd-dirty
I (58530) espnow_ota_responder: Running firmware version: v5.2.2
I (58537) esp_image: segment 0: paddr=001f0020 vaddr=3c020020 size=08c88h ( 35976) map
I (58549) esp_image: segment 1: paddr=001f8cb0 vaddr=3fc8ae00 size=0121ch (  4636) 
I (58554) esp_image: segment 2: paddr=001f9ed4 vaddr=40380000 size=06144h ( 24900) 
I (58565) esp_image: segment 3: paddr=00200020 vaddr=42000020 size=156bch ( 87740) map
I (58580) esp_image: segment 4: paddr=002156e4 vaddr=40386144 size=04bach ( 19372) 
I (70611) espnow_ota_responder: The device starts to upgrade
I (75576) app_main: Progress of OTA update: 10%
I (79356) app_main: Progress of OTA update: 20%
I (83419) app_main: Progress of OTA update: 30%
I (86737) app_main: Progress of OTA update: 40%
I (90558) app_main: Progress of OTA update: 50%
I (94506) app_main: Progress of OTA update: 60%
I (99547) app_main: Progress of OTA update: 70%
I (103972) app_main: Progress of OTA update: 80%
I (107774) app_main: Progress of OTA update: 90%
I (112108) app_main: Progress of OTA update: 100%
I (112108) espnow_ota_responder: Write total_size: 197584, written_size: 197584, spend time: 41s
I (112112) espnow_ota_responder: Running firmware version: v1.0-202-ge8748dd-dirty
I (112120) espnow_ota_responder: Running firmware version: v5.2.2
I (112127) esp_image: segment 0: paddr=001f0020 vaddr=3c020020 size=0a248h ( 41544) map
I (112140) esp_image: segment 1: paddr=001fa270 vaddr=3fc8c600 size=01294h (  4756) 
I (112144) esp_image: segment 2: paddr=001fb50c vaddr=40380000 size=04b0ch ( 19212) 
I (112154) esp_image: segment 3: paddr=00200020 vaddr=42000020 size=18a1ch (100892) map
I (112171) esp_image: segment 4: paddr=00218a44 vaddr=40384b0c size=07968h ( 31080) 
```

augustpevko, 2024-07-05:
@lhespress Yes, it works. I made the same changes before

lhespress, 2024-07-05:
@augustpevko Thanks for your feedback, i'll release it.

closed 2024-07-05 by zhanzhaocheng