espressif/esp-now #127: Usb ip with two esp32's

maaamcube, 2024-09-06:
Hi everyone,
I don't know where i can answer my question because there is no discussion section.
I looking to establish wifi comunication between two esp32's, one connected to a PC to the software and another one to Stm32's(USB device).
Is it possible to send the Stm32's USB's to the PC using the esp now ? like usb ip
Thank's.

lhespress, 2024-09-06:
@maaamcube PTAL [get-start](https://github.com/espressif/esp-now/tree/master/examples/get-started) example which show two  ESP32 comunication. One device receives data through the serial port and broadcasts it to all nodes through ESP-NOW data. The others will output the data through the serial port when receive the data.

lhespress, 2024-09-09:
@maaamcube PTAL [usb_wireless_bridge](https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-usb-bridge/examples/usb_wireless_bridge#wireless-mode) solutions.

maaamcube, 2024-09-16:
@lhespress i can't use usb for the MCU

lhespress, 2024-09-19:
@maaamcube The struct is as follows, right?
[image]

maaamcube, 2024-09-19:
@lhespress exactly, only usb, no uart.

lhespress, 2024-09-19:
@maaamcube  Correct?
[image]

maaamcube, 2024-09-19:
@lhespress yes

lhespress, 2024-09-19:
@maaamcube https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb

maaamcube, 2024-09-19:
@lhespress it's an example for usb only, there is no wifi.

lhespress, 2024-09-19:
@maaamcube First, confirm the USB function works fine based on usb example, Then replace UART to USB and confirm the ESP-NOW function works fine based on get-start example.

maaamcube, 2024-09-19:
@lhespress i tested https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb and https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_basic works fine but i don't know how to replace UART by USB, can you show me what i have to do ?

lhespress, 2024-11-01:
@maaamcube Please change [app_uart_initialize](https://github.com/espressif/esp-now/blob/master/examples/get-started/main/app_main.c#L66) API to USB.

closed 2024-12-06 by lhespress