espressif/esp-now #130: How do I pair devices with a specified mac address?

WzxKB, 2024-11-01:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

When studying the example of this component, I only found the pairing via broadcast in the example control etc., but what I want to do is a master device to search for the surrounding devices, display the information of the surrounding devices, and then be able to select a device as the pairing device function, if so, the pairing function needs a parameter of the other mac address. What should I do, please?

lhespress, 2024-11-01:
@WzxKB 

- You can call [espnow_send](https://github.com/espressif/esp-now/blob/master/src/espnow/include/espnow.h#L216) as unicast if you know the device's mac addr.

- You can reference [control](https://github.com/espressif/esp-now/tree/master/examples/control) example which only send data to the bind device if you don't know the device's mac addr.

dpnebert, 2024-11-21:
I'm going to piggyback on this because my question is an extension of the OP.

@WzxKB If you look at the control example, you'll see the "initiator" has "binding" code.  If you double click the BOOT button, it will temporarily "pair" with the "responder".  If the unit is power cycled, it will need to "bind" again.

Looking at the method, "static void app_espnow_event_handler(void *handler_args, esp_event_base_t base, int32_t id, void *event_data)", there is:
```
espnow_ctrl_bind_info_t *info = (espnow_ctrl_bind_info_t *)event_data;
```
That struct contains the "initiators' MAC Address.  If you save that to NVS, and at boot, load it and add it as a peer with "esp_now_add_peer"

----------------------------

None of the binding information is on the ESP-NOW programming guide, so I'll just have to read the source to figure it out.

----------------------------

Having said that, the method for sending is:
```
static void app_initiator_send_press_cb(void *arg, void *usr_data)
{
    static bool status = 0;

    ESP_ERROR_CHECK(!(BUTTON_SINGLE_CLICK == iot_button_get_event(arg)));

    if (s_espnow_ctrl_status == APP_ESPNOW_CTRL_BOUND) {
        ESP_LOGI(TAG, "initiator send press");
        espnow_ctrl_initiator_send(ESPNOW_ATTRIBUTE_KEY_1, ESPNOW_ATTRIBUTE_POWER, status);
        status = !status;
    } else {
        ESP_LOGI(TAG, "please double click to bind the devices firstly");
    }
}
```
Adding the initiator to the responder's peer list means that it isn't bound.  When I tap the BOOT button to "send press", the terminal says, "app_main: please double click to bind the devices firstly"

BUT!  If I comment out the conditional, leaving:
```
static void app_initiator_send_press_cb(void *arg, void *usr_data)
{
    static bool status = 0;

    ESP_ERROR_CHECK(!(BUTTON_SINGLE_CLICK == iot_button_get_event(arg)));

    if (s_espnow_ctrl_status == APP_ESPNOW_CTRL_BOUND) {
        ESP_LOGI(TAG, "initiator send press");
        espnow_ctrl_initiator_send(ESPNOW_ATTRIBUTE_KEY_1, ESPNOW_ATTRIBUTE_POWER, status);
        status = !status;
    } else {
        ESP_LOGI(TAG, "please double click to bind the devices firstly");
    }
}
```
It will send.

My question: How do peers and bounded devices work together?  If I add a peer, why is it not automatically bound?

lhespress, 2024-12-16:
@dpnebert

- ```espnow_ctrl_initiator_send``` send ```ESPNOW_ADDR_BROADCAST``` messgae, so it'll also send if you comment out the conditional.
- The responder will not handle the data if don't receive ```ESPNOW_DATA_TYPE_CONTROL_BIND``` message because of the initiator's MAC address can't be find in its list.

closed 2025-02-08 by lhespress