espressif/esp-now #132: Esprainmaker y EspNow

vizocu, 2024-12-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

No es posible el funcionamiento simultaneo de estas dos funcionalidades, estoy equivocado?

lhespress, 2024-12-16:
@vizocu 

Please apply the patch as the attachment [fan.zip](https://github.com/user-attachments/files/18147430/fan.zip) which use direction param for check ```BIND(1) and UNBIND(0)```, power for control the responder on [esp-rainmaker](https://github.com/espressif/esp-rainmaker/commit/7d506420d320a137bce6aaeb69a2059cc162efac).

[fan](https://github.com/espressif/esp-rainmaker/tree/master/examples/fan) log:
```
I (25956) app_main: Received write request via : Cloud
I (25966) app_main: this device is not been bound
I (25966) esp_rmaker_param: Reporting params: {"Fan":{"Direction":0}}
I (31596) esp_rmaker_param: Received params: {"Fan":{"Direction":1}}
I (31596) esp_rmaker_param: Found 1 params in write request for Fan
I (31596) app_main: Received write request via : Cloud
I (31606) app_main: initiator bind press
I (31616) esp_rmaker_param: Reporting params: {"Fan":{"Direction":1}}
I (33056) esp_rmaker_time: SNTP Synchronised.
I (33056) esp_rmaker_time: The current time is: Mon Dec 16 17:15:05 2024 +0800[CST], DST: No.
I (33686) esp_rmaker_param: Received params: {"Fan":{"Power":true}}
I (33686) esp_rmaker_param: Found 1 params in write request for Fan
I (33686) app_main: Received write request via : Cloud
I (33686) app_main: Received value = true for Fan - Power
I (33696) app_main: initiator send press
I (33706) esp_rmaker_param: Reporting params: {"Fan":{"Power":true}}
I (44896) esp_rmaker_param: Received params: {"Fan":{"Power":false}}
I (44896) esp_rmaker_param: Found 1 params in write request for Fan
I (44906) app_main: Received write request via : Cloud
I (44906) app_main: Received value = false for Fan - Power
I (44916) app_main: initiator send press
I (44916) esp_rmaker_param: Reporting params: {"Fan":{"Power":false}}
I (46336) esp_rmaker_param: Received params: {"Fan":{"Power":true}}
I (46336) esp_rmaker_param: Found 1 params in write request for Fan
I (46336) app_main: Received write request via : Cloud
I (46346) app_main: Received value = true for Fan - Power
I (46346) app_main: initiator send press
I (46356) esp_rmaker_param: Reporting params: {"Fan":{"Power":true}}
I (62006) esp_rmaker_param: Received params: {"Fan":{"Power":false}}
I (62006) esp_rmaker_param: Found 1 params in write request for Fan
I (62006) app_main: Received write request via : Cloud
I (62006) app_main: Received value = false for Fan - Power
I (62016) app_main: initiator send press
I (62026) esp_rmaker_param: Reporting params: {"Fan":{"Power":false}}
I (65176) esp_rmaker_param: Received params: {"Fan":{"Power":true}}
I (65176) esp_rmaker_param: Found 1 params in write request for Fan
I (65176) app_main: Received write request via : Cloud
I (65186) app_main: Received value = true for Fan - Power
I (65186) app_main: initiator send press
I (65196) esp_rmaker_param: Reporting params: {"Fan":{"Power":true}}
I (67016) esp_rmaker_param: Received params: {"Fan":{"Power":false}}
I (67016) esp_rmaker_param: Found 1 params in write request for Fan
I (67026) app_main: Received write request via : Cloud
I (67026) app_main: Received value = false for Fan - Power
I (67036) app_main: initiator send press
I (67036) esp_rmaker_param: Reporting params: {"Fan":{"Power":false}}
I (68346) esp_rmaker_param: Received params: {"Fan":{"Power":true}}
I (68356) esp_rmaker_param: Found 1 params in write request for Fan
I (68356) app_main: Received write request via : Cloud
I (68356) app_main: Received value = true for Fan - Power
I (68366) app_main: initiator send press
I (68376) esp_rmaker_param: Reporting params: {"Fan":{"Power":true}}
I (72756) esp_rmaker_param: Received params: {"Fan":{"Direction":0}}
I (72756) esp_rmaker_param: Found 1 params in write request for Fan
I (72756) app_main: Received write request via : Cloud
I (72766) app_main: initiator unbind press
I (72766) esp_rmaker_param: Reporting params: {"Fan":{"Direction":0}}
I (75516) esp_rmaker_param: Received params: {"Fan":{"Direction":1}}
I (75516) esp_rmaker_param: Found 1 params in write request for Fan
I (75526) app_main: Received write request via : Cloud
I (75526) app_main: initiator bind press
I (75536) esp_rmaker_param: Reporting params: {"Fan":{"Direction":1}}
I (77576) esp_rmaker_param: Received params: {"Fan":{"Power":false}}
I (77576) esp_rmaker_param: Found 1 params in write request for Fan
[... 21 lines cut ...]
I (81066) esp_rmaker_param: Reporting params: {"Fan":{"Power":true}}
```

responder [control](https://github.com/espressif/esp-now/tree/master/examples/control)

```
I (732) ESPNOW: espnow [version: 1.0] init
I (732) espnow: mac: c8:f0:9e:35:97:ac, version: 2
I (742) espnow: Enable main task
I (742) espnow: main task entry
I (742) app_main: primary 1 second 0
I (752) app_main: second 0 primary 8 
I (752) main_task: Returned from app_main()
I (3972) app_main: app_responder_ctrl_data_cb, initiator_attribute: 513, responder_attribute: 1, value: 0
I (7132) app_main: app_responder_ctrl_data_cb, initiator_attribute: 513, responder_attribute: 1, value: 1
I (8972) app_main: app_responder_ctrl_data_cb, initiator_attribute: 513, responder_attribute: 1, value: 0
I (10302) app_main: app_responder_ctrl_data_cb, initiator_attribute: 513, responder_attribute: 1, value: 1
I (14702) app_main: unbind, uuid: 30:30:f9:fb:8b:7c, initiator_type: 513
I (17472) espnow_ctrl: bind, timestamp: 17472, max timestamp: 30742, rssi: -19, min rssi: -55, bindlist size: 0
I (17472) app_main: bind, uuid: 30:30:f9:fb:8b:7c, initiator_type: 513
I (19542) app_main: app_responder_ctrl_data_cb, initiator_attribute: 513, responder_attribute: 1, value: 0
I (21162) app_main: app_responder_ctrl_data_cb, initiator_attribute: 513, responder_attribute: 1, value: 1
I (22202) app_main: app_responder_ctrl_data_cb, initiator_attribute: 513, responder_attribute: 1, value: 0
I (23002) app_main: app_responder_ctrl_data_cb, initiator_attribute: 513, responder_attribute: 1, value: 1
```

vizocu, 2024-12-17:
Gracias por contestar, no logro comprender la solución que me aportas. Podrías explicarmelo de otra forma para que pueda comprenderlo

lhespress, 2024-12-18:
@vizocu Could you provide you project links or your project as the attachment?

vizocu, 2024-12-18:
Claro, por un lado uso en un esp32 el código normal de un pulsador donde al pulsar envia un mensaje haciendo uso de la librería tinyespnow.h (esta parte funciona)
el ejemplo 1st
`#include "tinyESPNow.h"
static tinyESPNow tEN;

#define BUFSIZE (64)            // The size of the buffer you want to send and receive (MAX 249 Byte)

void dataUpdate() {
  uint8_t *buf = tEN.get();     // Get a pointer to the buffer
  buf[0] = 0xFF;
  Serial.printf("%02X=>", buf[1]);
  buf[1] = (buf[1] + 1) % 256;  // Add One to the received data
  Serial.printf("%02X\n", buf[1]);
  tEN.put(buf, 2);              // Update the buffer
  tEN.send();                   // Send the buffer
}

void setup() {
  uint8_t myId = 0xFF;          // Register your number
  Serial.begin( 115200 ); while (!Serial);
  tEN.begin(myId, BUFSIZE);
  dataUpdate();
}

void loop() {
  if (tEN.isRecv()) {           // Did the data arrive?
    dataUpdate();
  }
  tEN.update();                 // Update class
  delay(10);
`

Por otro lado uso otro esp32 donde quiero insertar en el ejemplo Rmakerswitch el ejemplo 2th de tinyespnow.
`// This example demonstrates the ESP RainMaker with a standard Switch device.
#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"
#include "AppInsights.h"

#define DEFAULT_POWER_MODE true
const char *service_name = "PROV_1234";
const char *pop = "abcd1234";

// GPIO for push button
#if CONFIG_IDF_TARGET_ESP32C3 || CONFIG_IDF_TARGET_ESP32C6
static int gpio_0 = 9;
static int gpio_switch = 7;
#else
// GPIO for virtual device
static int gpio_0 = 0;
static int gpio_switch = 16;
#endif

/* Variable for reading pin status*/
bool switch_state = true;

// The framework provides some standard device types like switch, lightbulb,
// fan, temperaturesensor.
static Switch *my_switch = NULL;

// WARNING: sysProvEvent is called from a separate FreeRTOS task (thread)!
void sysProvEvent(arduino_event_t *sys_event) {
  switch (sys_event->event_id) {
    case ARDUINO_EVENT_PROV_START:
#if CONFIG_IDF_TARGET_ESP32S2
      Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on SoftAP\n", service_name, pop);
      WiFiProv.printQR(service_name, pop, "softap");
#else
      Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on BLE\n", service_name, pop);
      WiFiProv.printQR(service_name, pop, "ble");
[... 109 lines cut ...]
  Serial.printf("%02X=>", buf[1]);
  buf[1] = (buf[1] + 1) % 256;  // Add One to the received data
  Serial.printf("%02X\n", buf[1]);
  tEN.put(buf, 2);              // Update the buffer
  tEN.send();                   // Send the buffer
}

void setup() {
  uint8_t myId = 0xFF;          // Register your number
  Serial.begin( 115200 ); while (!Serial);
  tEN.begin(myId, BUFSIZE);
  dataUpdate();
}

void loop() {
  if (tEN.isRecv()) {           // Did the data arrive?
    dataUpdate();
  }
  tEN.update();                 // Update class
  delay(10);
}`
Estoy trabajando con arduino IDE.
Simplificando, necesito que funcione el protocolo ESPnow y ESPRainmaker juntos para que al cambiar el estado del switch en rainmaker, este esp32 sea capaz de enviar  un cambio de estado al otro esp32.

Gracias por tu paciencia.

closed 2026-01-04 by lhespress