espressif/esp-now #133: Is there a limit of maximum devices receiving broadcast ESPNOW message

wuyuanyi135, 2024-12-15:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I am working on a project of a LED control system (~2000 leds) in a 300 sqm room. I'm considering using esp now to broadcast messages to these nodes to control the LED. I have tested a few devices and the signal was good. Since all nodes will be in listening mode I guess having more device won't necessarily cause communication problem. Could espressif confirm what the pitfall or limit is, to sending from one node to many using esp now?

svdrummer, 2024-12-15:
Google is your friend. According to ESP-NOW documentation, while the maximum number of paired devices for unicast communication is typically 20, when using broadcast packets, theoretically any number of devices can receive the broadcast message as it is sent to all devices within range, meaning there is no strict limit on the number of receivers for a broadcast ESP-NOW packet.

wuyuanyi135, 2024-12-15:
> Google is your friend. According to ESP-NOW documentation, while the maximum number of paired devices for unicast communication is typically 20, when using broadcast packets, theoretically any [...]

Thanks. Yes I did find that broadcast has no theoretical limit. I would like to see if any has actually tried such an extreme case that really involve this many nodes.