espressif/esp-now #134: Setting up ESP NOW with "peerInfo.encrypt = true;" Force to have a working WiFi connection to a router.

fableman73, 2025-01-04:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expect the encrypt to work without a WIFI router.

### Actual behavior (suspected bug)

No data is transmitted or received over the ESP NOW

### Error logs or terminal output

```shell
Using ESP32

If I enter a valid wifi network then it works.
If I enters an invalid WiFI network it fails to setup and work.

(If I run without encrypted data I can make it work without a wifi network.)

It's like the SEP NOW encrypted force to have a working WiFi else it will not work.
```

### Steps to reproduce the behavior

Implementing encrypted functions without a wifi router.

### Project release version

Latest

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

ESP32

### Shell

other (details in Additional context)

lhespress, 2025-01-06:
@fableman73 What do you mean change [peer->encrypt=false](https://github.com/espressif/esp-idf/blob/master/examples/wifi/espnow/main/espnow_example_main.c#L331) to [peer->encrypt=true](https://github.com/espressif/esp-idf/blob/master/examples/wifi/espnow/main/espnow_example_main.c#L331) or [espnow_add_peer](https://github.com/espressif/esp-now/blob/master/src/espnow/src/espnow.c#L437) has LMK, it must connect with a router?

fableman73, 2025-01-06:
Yes. As soon as you using encrypt in ESP NOW it will not be able to work correctly if you not connecting to a WiFi same time
 , adding peer is ok and everything but the communication fails.

fableman73, 2025-01-12:
Anyone manage to use encrypt messages between devices using ESP NOW without a connected WiFi router?

lhespress, 2026-01-04:
@fableman73 

ESP-NOW encryption does not require a WiFi router connection. The LMK/PMK encryption is peer-to-peer.

1. Both sender and receiver have the same PMK set via `esp_now_set_pmk()`
2. Both devices have matching LMK configured in `peer_info.lmk`

```
   esp_wifi_set_mode(WIFI_MODE_STA);
   esp_wifi_start();
   esp_now_init();
   esp_now_set_pmk(pmk);
   esp_now_peer_info_t peer = {
       .encrypt = true,
       .lmk = {...},
   };
   esp_now_add_peer(&peer);
```

lhespress, 2026-01-30:
@fableman73 Closing this issue since there has been no update on this. Please feel free to reopen if required