espressif/esp-now #136: ESP-NOW get peer num says 1 after fresh flash/boot, fetch error: Failed to fetch peer, error code = 12393

dpnebert, 2025-01-21:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Keeping in mind the 'get-started' example doesn't contain for adding for removing peers from the peer list.  Starting with the ESP-NOW 'get-started' example.  I added code to the end of app_main to log the number peers in the list:
```
.
.
.
espnow_set_config_for_data_type(ESPNOW_DATA_TYPE_DATA, true, app_uart_write_handle);

esp_now_peer_num_t num;
esp_err_t ret = esp_now_get_peer_num(&num);
if(ESP_OK != ret) {
    ESP_LOGE(TAG, "esp_now_get_peer_num failed, error code = %d", ret);
} else {
    ESP_LOGI(TAG, "esp_now_get_peer_num success, peer num = %d", num.total_num);
}
```
I expected to see:
```
I (651) espnow: main task entry
I (651) app_main: esp_now_get_peer_num success, peer num = 0
I (651) main_task: Returned from app_main()
```

### Actual behavior (suspected bug)

```
I (651) espnow: main task entry
I (651) app_main: esp_now_get_peer_num success, peer num = 1
I (651) main_task: Returned from app_main()
```

### Error logs or terminal output

```shell
I (31) boot: ESP-IDF v5.4 2nd stage bootloader
I (31) boot: compile time Jan 21 2025 14:38:13
I (31) boot: Multicore bootloader
I (32) boot: chip revision: v3.0
I (35) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : DIO
I (42) boot.esp32: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=17130h ( 94512) map
I (122) esp_image: segment 1: paddr=00027158 vaddr=3ff80000 size=0001ch (    28) load
I (122) esp_image: segment 2: paddr=0002717c vaddr=3ffb0000 size=03f28h ( 16168) load
I (132) esp_image: segment 3: paddr=0002b0ac vaddr=40080000 size=04f6ch ( 20332) load
I (141) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=74448h (476232) map
I (304) esp_image: segment 5: paddr=000a4470 vaddr=40084f6c size=13604h ( 79364) load
I (347) boot: Loaded app from partition at offset 0x10000
I (347) boot: Disabling RNG early entropy source...
I (358) cpu_start: Multicore app
I (366) cpu_start: Pro cpu start user code
I (366) cpu_start: cpu freq: 160000000 Hz
[... 104 lines cut ...]
    }
    free(peer_info);
```

And the output is:
```
I (3010) main: Number of peers: 1
I (3010) main: Number of peers: 1
E (3010) main: Failed to fetch peer, error code = 12393
```

'esp_now_fetch_peer':

What more confusing, the ESP-IDF API for ESP-NOW ([https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html)), describes 'esp_now_fetch_peer' ([https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html#_CPPv418esp_now_fetch_peerbP19esp_now_peer_info_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html#_CPPv418esp_now_fetch_peerbP19esp_now_peer_info_t)).  If you look at the ESP-IDF repo, it only contains a method signature in espnow.h [https://github.com/espressif/esp-idf/blob/0f0068fff3ab159f082133aadfa9baf4fc0c7b8d/components/esp_wifi/include/esp_now.h#L317](https://github.com/espressif/esp-idf/blob/0f0068fff3ab159f082133aadfa9baf4fc0c7b8d/components/esp_wifi/include/esp_now.h#L317).  But no mention of it in the ESP-NOW repo.

'esp_now_send':

What even more confusing, and starting become frustrating, is that the ESP-IDF Programming Guide describes esp_now_send [https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html#_CPPv412esp_now_sendPK7uint8_tPK7uint8_t6size_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html#_CPPv412esp_now_sendPK7uint8_tPK7uint8_t6size_t) as:
```
[esp_err_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_err.html#_CPPv49esp_err_t) esp_now_send(const uint8_t *peer_addr, const uint8_t *data, size_t len)
```

Which looking at the ESP-NOW repo, you won't find a definition for 'esp_now_send', but instead it has 'espnow_send'.  BTW, I have found the methods described in the ESP-NOW repo work, and those described in the ESP-IDF Programming Guide do not work.

And one last thing, I have looked for 'espnow_get_peer_num' incase like all the rest, if using the version without the underscore between 'esp' and 'now' works.

dpnebert, 2025-02-07:
Just to bump the table a bit so this doesn't get marked stale without anyone putting eyes on it.

I've completed everything except saving and loading the peer list to storage.  Fun fact, the ESP-NOW API guide fails to even mention 'espnow_storage_set'.  Which manually saving and loading a peer list isn't ideal or what I expected to have to do when I first started using ESP-NOW for my project.  I really expected to have a "save_bindlist_to_storage" and "load_bindlist_from_storage".  Maybe there is, but the documentation is lacking too much.  So I had to ditch binding and handle peers myself.

Sure hope that phantom peer doesn't cause issues.

lhespress, 2025-02-12:
@dpnebert 

The number of peers is `1` because of add broadcast address in [espnow_init](https://github.com/espressif/esp-now/blob/master/src/espnow/src/espnow.c#L1090) as default.

dpnebert, 2025-02-12:
@lhespress 

My receiver has a "learn task" that starts after a button is held down, and during this short window, if a transmitter within a certain RSSI range pushes a button, the receiver adds the transmitter's MAC to the peer list.  So, if I didn't want a broadcast, I could remove the peer with, "0xFF:0xFF:0xFF:0xFF:0xFF:0xFF" ?

Since we are on the subject, I first attempted to use binding, but didn't see a way to add bound peers to the peer list.  It really seemed like "binding" was for "one-time-use", and the user would have to rebind each time the power was cycled.

Is there any information about how to load/save the bind list of NVS?

lhespress, 2025-02-14:
@dpnebert Right. I think [control](https://github.com/espressif/esp-now/tree/master/examples/control) example match your requirement which provide `bind/unbind` function.

lhespress, 2025-02-24:
@dpnebert 

> Does the bind/unbinding list get saved in NVS? I had to rebind each time I rebooted.

Yes, see [espnow_ctrl_responder_set_bindlist](https://github.com/espressif/esp-now/blob/master/src/control/src/espnow_ctrl.c#L140) and [espnow_ctrl_responder_remove_bindlist](https://github.com/espressif/esp-now/blob/master/src/control/src/espnow_ctrl.c#L156C11-L156C48)

dpnebert, 2025-03-04:
@lhespress Thank you!  I had a reply typed last week, but like everyone, got pulled away.  If we could keep this open a bit longer, I will find a few moments to try it out.

If not, no worries, I will try to create one, but is there any examples of this in use?  When I looked on the ESP-NOW repo, I see the declaration and the implementation.  Now sure if there is an example in a different repo.

closed 2026-01-04 by lhespress