espressif/esp-now #138: Q: When using ESPNOW Broadcast, is it possible to experience 2-3 delays from the ESPNOW layer?

bhuvanchandra, 2025-02-19:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I would like to know the possible causes of delays when sending data via ESPNOW (broadcast) on both the TX and RX sides. In my understanding, there shouldn’t be much overhead that would introduce delays in the ESPNOW transmission pipeline.

lhespress, 2025-02-24:
@bhuvanchandra Yes, it maybe, below are the possible reason and how to improve it:

- The channel is congested (eg: other Wi-Fi networks, Bluetooth devices), it can cause delays in data transmission.
  - Use a less congested Wi-Fi channel (eg: 1, 6, 11)
- TX side sends data faster than the RX side can process it, the ESP-NOW queue on the RX side may overflow.
  - Implement flow control or rate limiting on the TX side
- Work on a power-saving mode (e.g., light sleep or modem sleep), it may introduce delays in waking up and processing incoming data.
  - Disable power-saving modes
- Packets are lost due to poor signal strength, it need retransmit the data.
  - Improve the signal strength by reducing the distance between devices or using external antennas.

bhuvanchandra, 2025-04-14:
@lhespress Thanks for your response.

We analyzed the delays and packet drops by adding a sequence number and timestamp to the ESP-NOW messages. We did not observe any delays. The minimal overhead is possibly on the serial data link we use to relay the ESP-NOW messages from the ESP32-C3 to the compute unit. Considering the PHY rate and the publish frequency we use, the air-time shall be negligible. 

Regarding the packet drops, we are investigating the possibility of packet collisions. I will open a separate thread for that discussion to avoid confusion. Thanks.

..............................................................

>The channel is congested (eg: other Wi-Fi networks, Bluetooth devices), it can cause delays in data transmission.
Use a less congested Wi-Fi channel (eg: 1, 6, 11)

We are operating in Japan. We are using channel 14 with 802.11b mode. My assumption is channel 14 will be less congested.

> TX side sends data faster than the RX side can process it, the queue on the RX side may overflow.
Implement flow control or rate limiting on the TX side

Noted.

> Work on a power-saving mode (e.g., light sleep or modem sleep), it may introduce delays in waking up and processing incoming data.
Disable power-saving modes

Yes, power-saving modes are disabled.

> Packets are lost due to poor signal strength, it need re-transmit the data.
Improve the signal strength by reducing the distance between devices or using external antennas.

In our use case we are using broadcast mode. No re-transmissions.

closed 2025-04-14 by bhuvanchandra