espressif/esp-now #139: Confusion about documentation

mtribiere, 2025-03-05:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello,
While trying to get started with ESP-NOW, I noticed that the documentation in this repo and the one in ESP-IDF are quite different. Even the function names between the examples in this repo and the reference in ESP-IDF are different.

As an example, in the example of this repo there is the `espnow_send()`, but in the ESP-IDF documentation the function is named `esp_now_send()`.

My question is: What resource should I trust? Maybe if there a purpose to those offsets?

Thank you for your help

lhespress, 2025-03-06:
@mtribiere You should reference ESP-IDF documentation if you use `esp_now_send` in your project. You should reference [ESP-NOW Programming Guide](https://docs.espressif.com/projects/esp-now/en/latest/esp32/index.html) if you use the component manager command add-dependency to add the esp-now to your project's dependency. This component provides some high-level functionalities to simplify the use of [ESP-NOW](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html) protocol from [ESP-IDF](https://github.com/espressif/esp-idf)

mtribiere, 2025-03-06:
Hi @lhespress thank you for the quick response!

If I understand this correctly, the `esp-now` package sits on top of `esp-idf`. It provides a higher level API for the ESP-NOW protocol implementation present in `esp-idf`.
It answers my question and this ticket can be closed.

closed 2025-03-06 by mtribiere