espressif/esp-now #142: Q: ESP-NOW: Does it follow 802.11 DCF behavior?

bhuvanchandra, 2025-04-14:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

While debugging the packet delays, we observed some packet drops, including occasional events of consecutive drops. In our use case, consecutive drops can lead to issues. We conducted a theoretical analysis, and based on the available information, ESP-NOW uses the IEEE 802.11 PHY and MAC layers, typically utilizing 802.11 action frames for communication.

From our understanding, 802.11 action frames follow CSMA/CA, which should drastically reduce the probability of collisions, especially given the small payload size (maximum 250 bytes). To reduce airtime, we increased the PHY data rate from the default 1 Mbps to 11 Mbps and above.

Additionally, by analyzing the Saturated and Unsaturated Bianchi models, we found that the MAC-layer drop probability is minimal. However, when increasing the RF Bit Error Rate (BER) to 10⁻⁴ or 10⁻³, we observed a higher drop rate.

Question:
Does ESP-NOW operate over IEEE 802.11 and therefore inherit the Distributed Coordination Function (DCF) behavior, which is the default MAC layer coordination mechanism in Wi-Fi?

zhangyanjiaoesp, 2025-04-14:
@bhuvanchandra 
you can use the Wireshark to capture the wifi packets. We need to know the actual scenario where the issue occurred and the packet capture.

The actual problem scenario includes: which chip was used? Which IDF version? Under what circumstances did the ESPNOW packet loss occur?

bhuvanchandra, 2025-04-14:
> [@bhuvanchandra](https://github.com/bhuvanchandra) you can use the Wireshark to capture the wifi packets. We need to know the actual scenario where the issue occurred and the packet capture.

Noted.

> The actual problem scenario includes: which chip was used? Which IDF version? 

Chip-set: ESP32-C3 (ESP32-C3-MINI-1)
ESP-IDF: v5.1.1

>  Under what circumstances did the ESPNOW packet loss occur?

The ESP32-C3 is installed on a mobile robots, which moves laterally. In a controlled setup, we have 2 stationary robots parked next to each other running test application to send-and-receive the data.

Channel: **14**
Mode: **802.11b**
Distance b/w 2 devices: roughly spaced around **1 meter**.
Max data payload: **250 Bytes**
Publish frequency: **5 Hz**
Tx power: **11 dBm**
PHY data Rate: **11 Mbps**

zhangyanjiaoesp, 2025-04-14:
@bhuvanchandra 
Does continuous packet loss occur only when two devices are sending ESP-NOW packets to each other? Are there any other devices nearby that are also transmitting Wi-Fi packets? Are ESP-NOW packets broadcast or unicast?

bhuvanchandra, 2025-04-14:
> [@bhuvanchandra](https://github.com/bhuvanchandra) Does continuous packet loss occur only when two devices are sending ESP-NOW packets to each other? Are there any other devices nearby that are also transmitting Wi-Fi packets? Are ESP-NOW packets broadcast or unicast?

In this setup, only 2 devices in channel 14. ESPNow packets are broadcast.

zhangyanjiaoesp, 2025-04-14:
@bhuvanchandra can you capture the wifi packets？

bhuvanchandra, 2025-04-15:
@zhangyanjiaoesp 
I could capture the Action frames. The MAC OUI for our device is: 02:52:4F

[image]

Please give me some time, we are running tests, will capture the packets and share the capture during the consecutive pkt drops.

bhuvanchandra, 2025-04-15:
@zhangyanjiaoesp could you share your email address, I cannot attach the capture file here in this thread.

zhangyanjiaoesp, 2025-04-15:
@bhuvanchandra 
You can upload your capture file [here](https://wormhole.app/) and share us the download link.

bhuvanchandra, 2025-04-15:
@zhangyanjiaoesp thanks... here is the capture file: https://wormhole.app/o49YYD#jVVyrRrZ1nnUt2Q_p_dxrQ
In this test the 2 devices are approximately 1 meter away. Both sending and receiving the data at 20Hz.

zhangyanjiaoesp, 2025-04-15:
@bhuvanchandra 
If you transmit on channels 1~13 with the default TX power set to 20 dBm, will this kind of problem occur?

bhuvanchandra, 2025-04-15:
> [@bhuvanchandra](https://github.com/bhuvanchandra) If you transmit on channels 1~13 with the default TX power set to 20 dBm, will this kind of problem occur?

@zhangyanjiaoesp We can test it. But not sure if it will help/improve.

zhangyanjiaoesp, 2025-04-16:
@bhuvanchandra 
Did you already apply any filter when capturing packets on channel 14? We prefer not to set any filters so that we can capture all packets in the air. This helps us determine whether interference from the surrounding environment is affecting the transmission and reception of ESP-NOW broadcast packets.

One more thing to confirm — when you're sending ESP-NOW data packets, is the device also engaged in any other communication (such as Bluetooth, or packet transmission between STA and AP)?

bhuvanchandra, 2025-04-16:
@zhangyanjiaoesp 
> Did you already apply any filter when capturing packets on channel 14? We prefer not to set any filters so that we can capture all packets in the air. This helps us determine whether interference [...]

I applied this filter: `wlan.fc.type_subtype == 13 && wlan.tag.oui == 0x18fe34 && (wlan.sa[0:3] == 02:52:4f || wlan.sa[0:3] == 04:52:4F || wlan.sa[0:3] == 06:52:4F)`

I understand your point. Will get unfiltered action frames. Will apply only `wlan.fc.type_subtype == 13` 

> One more thing to confirm — when you're sending ESP-NOW data packets, is the device also engaged in any other communication (such as Bluetooth, or packet transmission between STA and AP)?

No other communication, only ESPNow.

zhangyanjiaoesp, 2025-04-16:
> I understand your point. Will get unfiltered action frames. Will apply only `wlan.fc.type_subtype == 13`

@bhuvanchandra This is not needed either. Do not use any filter!

bhuvanchandra, 2025-04-16:
> > I understand your point. Will get unfiltered action frames. Will apply only `wlan.fc.type_subtype == 13`
> 
> [@bhuvanchandra](https://github.com/bhuvanchandra) This is not needed either. Do not use any filter!

Okay

bhuvanchandra, 2025-04-29:
@zhangyanjiaoesp apologies for the delay. While doing component isolated tests we observed some packet drops due to serial data-link. We are doing more testes and fixing the serial data-link pipeline before using the esp-now. Will get back to you soon. Thanks for your time and support.

lhespress, 2026-01-04:
@bhuvanchandradv Closing this issue since there has been no update on this. Please feel free to reopen if required.

bhuvanchandradv, 2026-01-05:
> [@bhuvanchandradv](https://github.com/bhuvanchandradv) Closing this issue since there has been no update on this. Please feel free to reopen if required.

@lhespress 
Sorry—I overlooked updating this. Based on the investigation, the bottleneck is the UART data-link pipeline. We optimized it on both the ESP32 firmware side and the Linux application side. Up to 500–600 Hz RX traffic, packet drops are now minimal. Thanks.