espressif/esp-now #145: error: passing argument 3 of 'uxTaskGetSystemState' from incompatible pointer type [-Wincompatible-pointer-types]

franz-ms-muc, 2025-06-26:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

ESP-IDF 5.4.1
  espressif/esp-now:  version: 2.5.2

```
/home/franz/git/hr/LOGIBUS_integration_HutschienenmopedXL/Application/managed_components/espressif__esp-now/src/utils/src/espnow_mem.c: In function 'espnow_mem_print_task':
/home/franz/git/hr/LOGIBUS_integration_HutschienenmopedXL/Application/managed_components/espressif__esp-now/src/utils/src/espnow_mem.c:148:72: error: passing argument 3 of 'uxTaskGetSystemState' from incompatible pointer type [-Wincompatible-pointer-types]
  148 |     uxArraySize = uxTaskGetSystemState(pxTaskStatusArray, uxArraySize, &ulTotalRunTime);
      |                                                                        ^~~~~~~~~~~~~~~
      |                                                                        |
      |                                                                        uint32_t * {aka long unsigned int *}

```

franz-ms-muc, 2025-07-08:
@igrr is nobody monitoring this Repo ?

lhespress, 2026-01-04:
@franz-ms-muc It'll fix it if select `configRUN_TIME_COUNTER_TYPE` to uint32_t.

closed 2026-01-06 by zhanzhaocheng