espressif/esp-now #150: Does not compile with ESP-IDF v5.5

finger563, 2025-08-21:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Compile when building with ESP-IDF 5.5

### Actual behavior (suspected bug)

Fails to compile.

1. missing `REQUIRES` for `esp_driver_gpio`, `esp_driver_uart`, and `esp_driver_usb_serial_jtag`
2. Has incompatible function pointer type:

```console
/Users/bob/finger563/esp-now/src/espnow/src/espnow.c:1086:46: error: passing argument 1 of 'esp_now_register_send_cb' from incompatible pointer type [-Wincompatible-pointer-types]
 1086 |     ESP_ERROR_CHECK(esp_now_register_send_cb(espnow_send_cb));
      |                                              ^~~~~~~~~~~~~~
      |                                              |
      |                                              void (*)(const uint8_t *, esp_now_send_status_t) {aka void (*)(const unsigned char *, esp_now_send_status_t)}
/Users/bob/esp/esp-idf/components/esp_common/include/esp_err.h:116:30: note: in definition of macro 'ESP_ERROR_CHECK'
  116 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
In file included from /Users/bob/finger563/esp-now/src/espnow/src/espnow.c:18:
/Users/bob/esp/esp-idf/components/esp_wifi/include/esp_now.h:185:54: note: expected 'esp_now_send_cb_t' {aka 'void (*)(const wifi_tx_info_t *, esp_now_send_status_t)'} but argument is of type 'void (*)(const uint8_t *, esp_now_send_status_t)' {aka 'void (*)(const unsigned char *, esp_now_send_status_t)'}
  185 | esp_err_t esp_now_register_send_cb(esp_now_send_cb_t cb);
      |                                    ~~~~~~~~~~~~~~~~~~^~
```

### Error logs or terminal output

```shell
/Users/bob/finger563/esp-now/src/espnow/src/espnow.c:1086:46: error: passing argument 1 of 'esp_now_register_send_cb' from incompatible pointer type [-Wincompatible-pointer-types]
 1086 |     ESP_ERROR_CHECK(esp_now_register_send_cb(espnow_send_cb));
      |                                              ^~~~~~~~~~~~~~
      |                                              |
      |                                              void (*)(const uint8_t *, esp_now_send_status_t) {aka void (*)(const unsigned char *, esp_now_send_status_t)}
/Users/bob/esp/esp-idf/components/esp_common/include/esp_err.h:116:30: note: in definition of macro 'ESP_ERROR_CHECK'
  116 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
In file included from /Users/bob/finger563/esp-now/src/espnow/src/espnow.c:18:
/Users/bob/esp/esp-idf/components/esp_wifi/include/esp_now.h:185:54: note: expected 'esp_now_send_cb_t' {aka 'void (*)(const wifi_tx_info_t *, esp_now_send_status_t)'} but argument is of type 'void (*)(const uint8_t *, esp_now_send_status_t)' {aka 'void (*)(const unsigned char *, esp_now_send_status_t)'}
  185 | esp_err_t esp_now_register_send_cb(esp_now_send_cb_t cb);
      |                                    ~~~~~~~~~~~~~~~~~~^~
```

### Steps to reproduce the behavior

1. Use ESP-IDF 5.5
2. try to build one of the examples (in this case the `switch` example)

### Project release version

`master` branch

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

MacOS

### Shell

ZSH

lhespress, 2026-01-04:
@finger563 Since we have verified that the issue has been fixed in master branch, let’s close it. Please feel free to reopen it if the problem still occurs on your side.

closed 2026-01-04 by lhespress

WzxKB, 2026-01-29:
> [@finger563](https://github.com/finger563) Since we have verified that the issue has been fixed in master branch, let’s close it. Please feel free to reopen it if the problem still occurs on your side.

这个虽然在最新的master分支更新了，但是我们使用的ESP Component Registry里面没有更新，那里依然是V2.5.2版本